vilemduha/blendercam #201: Beginner's question: "use exact mode"

Karsten68, 2022-02-02:
Hi.
I am struggling a bit after installing blendercam with opencamlib on blender 2.9.2., Debian 10 Machine.
It's probably just a "user error": 
I see from the documentation, that the "use exact mode" setting is described and shown in the "CAM optimisation" panel. However, it seems to be missing on my display? What am I doing wrong?

[image]

abosafia, 2022-02-03:
Hi.
Nothing missed. It doesn't work without exact mode. That's why it isn't there to choose.

Karsten68, 2022-02-03:
Thanks for the quick answer.

closed 2022-02-03 by Karsten68