vilemduha/blendercam #202: Scale problem (and a hacked fix)

imerso, 2022-02-18:
So, as probably many are suffering with this bug, where you set the scene scale to metric, millimeters and 0.001 scale -- which makes sense because you want to work in millimeters -- but then get exported gcode with huge numbers...

What worked for me (not sure but maybe you will need different number here) was:

gcodepath.py line 141:
```
        unitcorr = 0.1;
```

gcodepath.py line 240:
```
            c.write('G00 Z' + str(fmh) + '\n')
```

Well, as said, your number might be different, but these two changes solved the problem for me and now I'm getting proper gcode scale.

I may come back to this in the future, and make a definitive fix. But for now it is enough for me. =)

Good luck.

imerso, 2022-02-18:
Well, it's quite buggy... but I like it and will see if I manage to reserve free time to spend fixing some of them.

abosafia, 2022-03-27:
We never had this issue.
please join us at
https://riot.im/app/#/room/#blendercam:matrix.org

fl0wm0ti0n, 2022-08-21:
i run into this... i have no clue... everything is correct in mm sizes... but the gcode output is x10 bigger...

[image]
[image]

[Op_Cube_1.gcode.txt](https://github.com/vilemduha/blendercam/files/9389219/Op_Cube_1.gcode.txt)

fl0wm0ti0n, 2022-08-21:
> i run into this... i have no clue... everything is correct in mm sizes... but the gcode output is x10 bigger...
> 
> [image] [image]
> 
> [Op_Cube_1.gcode.txt](https://github.com/vilemduha/blendercam/files/9389219/Op_Cube_1.gcode.txt)

Ok the solution for me was, to start a new project and from begining, set the settings to Millimeters and dont set the scaling to 0.01mm, let it on 1mm. in viewport set it to 0.01

closed 2023-01-14 by pppalain