vilemduha/blendercam #208: BAS relief as standalone script

sandorkonya, 2022-04-04:
@pppalain - i saw that you contributed to the BAS-relief code.
I have very limited understanding of blenders internal structure, but as far i understand, the script takes an input image (a height/depth map) and creates a BAS relief for it, right?
If yes, could this script be turned into a standalone function to have the depthmap as input and output the resulting BAS relief itself as a depth map image? 
Thank you for your time.

closed 2022-04-04 by sandorkonya

sandorkonya, 2022-04-05:
@vilemduha thank you for your reply.
... I was not sure that it makes sense what i ask.

I will try to fiddle around with the code.

abosafia, 2022-04-06:
> Yeah, even if you deleted the question for some reason, the whole addon can be easily separated from Blender - it's in blender just for convenience...

Welcome back blendercam master
Thanks for the great add-on

abosafia, 2022-04-06:
> @vilemduha thank you for your reply. ... I was not sure that it makes sense what i ask.
> 
> I will try to fiddle around with the code.

Join us at https://matrix.to/#/#blendercam:matrix.org