vilemduha/blendercam #222: [BUG]: Fails to generate gcode in blender 3.0.1

michael-barrow, 2022-08-08:
**Issue**
Blendercam fails to generate gcode for the waterline method with the following error:
```
Read blend: /home/king/Documents/CAD/Air_Filter/Bottom Panel.blend
Color management: scene view "Filmic" not found, setting default "Standard".
CALCULATE
OBJECT
update rest 
update rest 
update rest 
changing sampling resolution to 0.000137
3
valid geometry
minz from object:-0.05000000074505806
update rest 
material_from_model
progress{retrieving object slices}
progress{getting zbuffer}
fn-l: /home/king/Documents/CAD/Air_Filter/
bn: Bottom Panel
Python: Traceback (most recent call last):
  File "/opt/blender/blendercam/scripts/addons/cam/ops.py", line 214, in execute
    gcodepath.getPath(context, o)
  File "/opt/blender/blendercam/scripts/addons/cam/gcodepath.py", line 510, in getPath
    getPath3axis(context, operation)
  File "/opt/blender/blendercam/scripts/addons/cam/gcodepath.py", line 674, in getPath3axis
    prepareArea(o)
  File "/opt/blender/blendercam/scripts/addons/cam/image_utils.py", line 1214, in prepareArea
    renderSampleImage(o)
  File "/opt/blender/blendercam/scripts/addons/cam/image_utils.py", line 1091, in renderSampleImage
    r.tile_x = 1024  # ceil(resx/1024)
AttributeError: 'RenderSettings' object has no attribute 'tile_x'

location: <unknown location>:-1
Error: Python: Traceback (most recent call last):
  File "/opt/blender/blendercam/scripts/addons/cam/ops.py", line 214, in execute
    gcodepath.getPath(context, o)
  File "/opt/blender/blendercam/scripts/addons/cam/gcodepath.py", line 510, in getPath
    getPath3axis(context, operation)
  File "/opt/blender/blendercam/scripts/addons/cam/gcodepath.py", line 674, in getPath3axis
    prepareArea(o)
  File "/opt/blender/blendercam/scripts/addons/cam/image_utils.py", line 1214, in prepareArea
    renderSampleImage(o)
  File "/opt/blender/blendercam/scripts/addons/cam/image_utils.py", line 1091, in renderSampleImage
    r.tile_x = 1024  # ceil(resx/1024)
AttributeError: 'RenderSettings' object has no attribute 'tile_x'

location: <unknown location>:-1

```
It seems an incompatability issue with 3.0.1. Do I need to downgrade or is there some deeper problem? I found the same issue on Blender 3.2.x but that is not listed as compatible.

Cannot attach .blend file as it seems they are not supported but all dependencies are installed, see below for UI sate

[image]

**Steps to Reproduce**
I followed [this ](https://www.youtube.com/watch?v=ZxluRPKOUfA&ab_channel=Stefan)tutorial.

abosafia, 2022-08-08:
Hi Michael.
Water line doesn't work now because blender changed. Maybe later it'll be fixed.

michael-barrow, 2022-08-08:
Hmm...  @abosafia Ok looks like the fix for that particular error is something like [this](https://github.com/Naxela/The_Lightmapper/commit/e1e27ee6719473e5ad82f401563830970660c23f), and it also seems that image_utils.py is part of blendercam. So, if I wanted to try and patch this, would it be for this repo?

I might have some time over the weekend if you could point me at places things need to be touched.

ksuprynowicz, 2022-08-08:
If you'd like to generate waterlines with the current version, you will need to install opencamlib too. Generally toolpaths generated by BlenderCAM + opencamlib are of much better quality than BlenderCAM alone. If you fix waterlines without opencamlib, send us a pull request - it will be very welcome :)

abosafia, 2022-08-09:
Please join us 
https://riot.im/app/#/room/#blendercam:matrix.org

michael-barrow, 2022-08-09:
> If you'd like to generate waterlines with the current version, you will need to install opencamlib too. Generally toolpaths generated by BlenderCAM + opencamlib are of much better quality than [...]

Thanks @abosafia, I had installed opencamlib before trying. So I will take a look at the stack I have over the weekend as if I could get it working that seems it would give best results. Not sure from the previous message that this is a desirable pr or not but I will post any update when I have it here. Sorry I don't have a riot.im account.

Cheers!

ksuprynowicz, 2022-08-09:
You need to not only install opencamlib, but also enable exact mode and using opencamlib for a given operation. If messages like "getting zbuffer" show up, that means that opencamlib use is not enabled for a given operation.
As for the PR, it's most definitely desirable, but might be difficult considering that raster-based waterlines depend on now removed Blender renrerer.

michael-barrow, 2022-08-10:
@ksuprynowicz thanks for the tips! Actually I don't see the exact mode selection anywhere in the options, just that Opencamlib is installed... I have experimental features enabled and have looked over everything in the renderer options. Where exactly would it be?

michael-barrow, 2022-08-10:
@ksuprynowicz I see in riot.im that this actually is working (for some folks) but I am just not able to drive it correctly. I guess this can be closed?

Michael

ksuprynowicz, 2022-08-10:
I t

ksuprynowicz, 2022-08-10:
I tried everything I could, and it doesn't work for me. It does work with the provided example though. I'm pretty sure there's some bug there, because all the settings are the same. Opencamlib support for waterlines was removed recently, but I will try to bring it back soon. It doesn't have infill yet so it's mostly for finishing, but it's much faster and more accurate that raster-based algorithm. I tried rolling back the repo to before it was removed and it works perfectly.

abosafia, 2022-08-17:
Hi,
You can try this till it be merged.
https://github.com/pppalain/blendercam
Water line now uses ocl.
For the moment. It doesn't produce path without ocl.

closed 2023-01-14 by pppalain