vilemduha/blendercam #223: Update Blogspot website of blendercam

Franckapik, 2022-08-10:
Hi @vilemduha , @ksuprynowicz ,

I am the contributor who update the last readme on GitHub repository. 
I am working on webdesign and I am a big user of blendercam. 
I would like to contribute again to this project and I am desapointed to see that the most referenced website of blendercam on Google  seems old and deprecated. I think that we lost, like me at the beginning, people who think that blendercam is no longer supported. 
So that's why I was wondering if it was possible to have access in a way to the blogspot website and update it and/or redirect to a new website I can make in September with documentation updated to the last features. 

Thank you a lot for your attention,
Franckapik.

abosafia, 2022-08-10:
That's good idea.

ksuprynowicz, 2022-08-10:
Agreed. I will email Vilem about this, but I'm not sure if the email adress I have is his current one.

Franckapik, 2022-08-10:
Thank you really much @ksuprynowicz . 
I am waiting news from you hoping that the email address is the good one 🙂

ksuprynowicz, 2022-08-10:
If not, we could make a new website. Tux Family offers free hosting for open source projects - we use it in another open source project I'm contributing to, and it's awesome.
https://www.tuxfamily.org/
I can register a domain if we decide to do it.

Franckapik, 2022-08-10:
I like the idea to use tuxfamily for hosting a new website! I red that it could be possible to have another domain name than .tuxfamily.org . I don't know if it's really important but sometimes it could be easier to have a domain name like blendercam.com/org/io ... 

Don't hesitate to ask me if you need something to register the domain. 

I would add to major contributors to say here if something seems wrong to you in creating a new website in addition to the GitHub page ... 

Thank you!

Franckapik, 2023-01-01:
Hi @ksuprynowicz !

Happy new year 🥳 to you! 

Just to know ... Have you tried to contact  Vilem ? The blogspot website is really starting to be old...
And I read that even if the author has forgotten the password, he can ask for deleting the account... 
Working on the new website for 2023 ☺️

Franckapik, 2023-01-17:
I am going to create the domain name www.blendercam.com today.

[image]

The principal function of this "official" website will be to have the link to the github main repo, the link to join discussions, the  link to download the eventual zip file, and the current documentation.

@abosafia , @pppalain , @ksuprynowicz , @razcore-rad, @TurBoss, ...

**Can i have your your final go-ahead for this ?**

Thank you !
F.

TurBoss, 2023-01-17:
awesome :+1:

ksuprynowicz, 2023-01-17:
Sounds awesome to me :)

abosafia, 2023-01-17:
Very good, 👍
Thanks 🙏

narvf, 2023-01-17:
Great idea!

razcore-rad, 2023-01-17:
> I am going to create the domain name [www.blendercam.com](http://www.blendercam.com) today.
>
> [image]
>
> The principal function of this "official" website will be to have the link to the github main [...]

I haven't thought about it too much, but my project is probably going to be something completely different, just inspired by BlenderCAM at the end of the day. I don't think it's appropriate for me to take it, change it so much and still refer to it as BlenderCAM.

Since I'm basically constructing it from scratch I'll also change the workflow quite a bit for my particular needs and I won't have the man-power to do feature requests and customizations being just one person.

It may not be to others' taste if they come from BlenderCAM, but we'll have to see about that, I have a lot to do.

Franckapik, 2023-01-17:
@razcore-rad . I understand your position. 

The priority of the website is to federate all the collaborators and users towards the same place by reassuring on the idea that the addon is still open source and really active with mutual aid and discussions that we can see every day.
The hardest part is obviously to keep the documentation up to date but I will continue to work on it soon.
I clearly don't want to divide or put pressure on the people working on its development.

An open source project is always on the move with derived paths. Here, this is just about update and formalization to guide people in need to have a free tool for their machines. According to google searches, we still have the impression that the project stopped in 2015 :/

Thank you all for agreeing to this idea.

pppalain, 2023-12-14:
in the end of the day, no good came of this

Franckapik, 2024-01-02:
The blendercam website is now here on www.blendercam.com :)