vilemduha/blendercam #227: Blender 3.3 when calculate, snap_align keyword error

wcanarro, 2023-01-09:
Hi, I am trying for the first time BlenderCam and after added an operation I clicked "Calculate Path and Export GCode" button and I get the error:
Python: Traceback (most recent call last):
  File "C:\Program Files\blendercam-master\scripts\addons\cam\ops.py", line 214, in execute
    gcodepath.getPath(context, o)
  File "C:\Program Files\blendercam-master\scripts\addons\cam\gcodepath.py", line 541, in getPath
    getPath3axis(context, operation)
  File "C:\Program Files\blendercam-master\scripts\addons\cam\gcodepath.py", line 665, in getPath3axis
    chunks.extend(utils.sampleChunks(o, pathSamples, layers))
  File "C:\Program Files\blendercam-master\scripts\addons\cam\utils.py", line 370, in sampleChunks
    prepareBulletCollision(o)
  File "C:\Program Files\blendercam-master\scripts\addons\cam\collision.py", line 259, in prepareBulletCollision
    bpy.ops.transform.resize(value=(BULLET_SCALE, BULLET_SCALE, BULLET_SCALE),
  File "C:\Program Files\Blender Foundation\Blender 3.3\3.3\scripts\modules\bpy\ops.py", line 113, in __call__
    ret = _op_call(self.idname_py(), None, kw)
TypeError: Converting py args to operator properties: : keyword "snap_align" unrecognized

abosafia, 2023-01-10:
Hi, wcanarro.
Try this.
https://github.com/pppalain/blendercam

pppalain, 2023-01-14:
the pppalain branch has just now been changed to the main branch

closed 2023-01-14 by pppalain