vilemduha/blendercam #229: Generating Gcode for laser with parallel strategy

wcanarro, 2023-02-10:
Hi, I am getting the error showed in the picture
[image]
maybe I am missing something...I dont think there ia an issue...any help?

TurBoss, 2023-02-10:
hello, looks like the dial.002 was created outside the Scene Collection

can you move it to that collection and try again?

thanks

wcanarro, 2023-02-10:
yes, the error is gone and the gcode generated, but I dont see the result inside blender...and the mesh must be in the main directory (the scene collection and not in sub folder)

wcanarro, 2023-02-10:
Now, if I try on the same mesh, parallel strategy and select the laser (preset -> laser) i get this error
[image]

abosafia, 2023-02-24:
Laser cannot be used with 3d strategies like parallel.
It is for pocket and profile cut.

pppalain, 2023-02-24:
I consider the issue to be closed.  Training wheels will be added for people not to choose the parallel or drop cutter strategies to laser tool.

closed 2023-02-24 by pppalain