vilemduha/blendercam #232: not able to activate/use blendercam wtth blender 3.4.1

cruzmiester, 2023-03-11:
not sure if this the right way of reporting issues with software from Github.... if it isn't I apologize, if it is, I'm not able to activate Blendercam with Blender 3.4.1 using Windows 10 (from zip file download). If it matters the only addon I added previously is jewelcraft also hosted on Github.
this is the error I get when I try to activate it using your install instruction (or at least as I understand them):

Traceback (most recent call last):
  File "C:\Program Files\Blender Foundation\Blender 3.4\3.4\scripts\modules\addon_utils.py", line 333, in enable
    mod = __import__(module_name)
  File "D:\cruzmiester\Documents\Blender\Assets\blendercam-master\scripts\addons\cam\__init__.py", line 47, in <module>
    from cam import ui, ops, curvecamtools, curvecamequation, curvecamcreate, utils, simple, \
  File "D:\cruzmiester\Documents\Blender\Assets\blendercam-master\scripts\addons\cam\ui.py", line 33, in <module>
    from cam import gcodeimportparser, simple
  File "D:\cruzmiester\Documents\Blender\Assets\blendercam-master\scripts\addons\cam\simple.py", line 31, in <module>
    from shapely.geometry import Point, LineString, Polygon, MultiLineString
ModuleNotFoundError: No module named 'shapely'

abosafia, 2023-03-11:
Hi cruzmiester,
https://youtu.be/RXpGLJz7ZIs

RneeJ, 2023-03-12:
Same problem Blender 3.4.1 on Ubuntu 22.04

abosafia, 2023-03-13:
Hi all,
Join us at
https://matrix.to/#/#blendercam:matrix.org

pppalain, 2023-03-24:
so shapely was installed in the system python which is not the same as the blender python.  blender python thinks it is installed but it can't find it.  the easiest way is to uninstall shapely and other dependancies from system python and just let the script autoinstall dependencies.

AVHon, 2023-05-14:
I'm having the same error, but curiously, re-running the installer makes pip say that `shapely` is already installed, but then Blender can't find it.

```
C:\Program Files\blender>.\blender
Read prefs: C:\Users\me\AppData\Roaming\Blender Foundation\Blender\3.3\config\userpref.blend
Looking in links: c:\Users\me\AppData\Local\Temp\tmp49jwtjup
Requirement already satisfied: setuptools in c:\program files\blender\3.3\python\lib\site-packages (63.2.0)
Requirement already satisfied: pip in c:\program files\blender\3.3\python\lib\site-packages (22.3.1)
Requirement already satisfied: shapely in c:\users\me\appdata\roaming\python\python310\site-packages (2.0.1)
Requirement already satisfied: Equation in c:\users\me\appdata\roaming\python\python310\site-packages (1.2.1)
Requirement already satisfied: opencamlib in c:\users\me\appdata\roaming\python\python310\site-packages (2023.1.11)
Requirement already satisfied: numpy>=1.14 in c:\program files\blender\3.3\python\lib\site-packages (from shapely) (1.22.0)

[notice] A new release of pip available: 22.3.1 -> 23.1.2
[notice] To update, run: C:\Program Files\blender\3.3\python\bin\python.EXE -m pip install --upgrade pip
Traceback (most recent call last):
  File "C:\Program Files\blender\3.3\scripts\modules\addon_utils.py", line 333, in enable
    mod = __import__(module_name)
  File "C:\Users\me\SourceCode\blendercam\scripts\addons\cam\__init__.py", line 47, in <module>
    from cam import ui, ops, curvecamtools, curvecamequation, curvecamcreate, utils, simple, \
  File "C:\Users\me\SourceCode\blendercam\scripts\addons\cam\ui.py", line 34, in <module>
    from cam import gcodeimportparser, simple
  File "C:\Users\me\SourceCode\blendercam\scripts\addons\cam\simple.py", line 31, in <module>
    from shapely.geometry import Point, LineString, Polygon, MultiLineString
ModuleNotFoundError: No module named 'shapely'
```
I have the same problem whether following the instructions in [the documentation](https://github.com/vilemduha/blendercam/blob/master/documentation/Blendercam%20Installation.md) (normal Blender install, letting the addon call pip in order to install the dependencies) or in the above-linked [youtube video](https://youtu.be/RXpGLJz7ZIs) (portable copy of Blender, manually installing the dependencies by running Blender's pip in the command line).

In both cases, I'm using Blender 3.3.6 (the most recent LTS release).

pppalain, 2023-05-14:
so try pip uninstall shapely and let the script install it.  you see there is the system shapely and the blender python's shapely.  it thinks it's satisfied but it is installed in the system.  but blender cannot access the system's shapely.

AVHon, 2023-05-16:
There is no system copy of Python, but I found the packages installed in `C:\Users\me\AppData\Roaming\Python\Python310\site-packages`. I deleted all of the files in that directory, and re-did the installation process (as described in the documentation, with a normal installation of Blender and BlenderCAM calling Blender's copy of pip).

```
Read prefs: C:\Users\me\AppData\Roaming\Blender Foundation\Blender\3.3\config\userpref.blend
Defaulting to user installation because normal site-packages is not writeable
Looking in links: c:\Users\me\AppData\Local\Temp\tmpdvutu0ft
Requirement already satisfied: setuptools in c:\program files\blender foundation\blender 3.3\3.3\python\lib\site-packages (63.2.0)
Requirement already satisfied: pip in c:\program files\blender foundation\blender 3.3\3.3\python\lib\site-packages (22.3.1)
Defaulting to user installation because normal site-packages is not writeable
Collecting shapely
  Using cached shapely-2.0.1-cp310-cp310-win_amd64.whl (1.4 MB)
Collecting Equation
  Using cached Equation-1.2.01.zip (25 kB)
  Preparing metadata (setup.py) ... done
Collecting opencamlib
  Using cached opencamlib-2023.1.11-cp310-cp310-win_amd64.whl (272 kB)
Requirement already satisfied: numpy>=1.14 in c:\program files\blender foundation\blender 3.3\3.3\python\lib\site-packages (from shapely) (1.22.0)
Installing collected packages: Equation, shapely, opencamlib
  DEPRECATION: Equation is being installed using the legacy 'setup.py install' method, because it does not have a 'pyproject.toml' and the 'wheel' package is not installed. pip 23.1 will enforce this behaviour change. A possible replacement is to enable the '--use-pep517' option. Discussion can be found at https://github.com/pypa/pip/issues/8559
  Running setup.py install for Equation ... done
Successfully installed Equation-1.2.1 opencamlib-2023.1.11 shapely-2.0.1

[notice] A new release of pip available: 22.3.1 -> 23.1.2
[notice] To update, run: C:\Program Files\Blender Foundation\Blender 3.3\3.3\python\bin\python.EXE -m pip install --upgrade pip
Traceback (most recent call last):
  File "C:\Program Files\Blender Foundation\Blender 3.3\3.3\scripts\modules\addon_utils.py", line 333, in enable
    mod = __import__(module_name)
  File "C:\Users\me\SourceCode\blendercam\scripts\addons\cam\__init__.py", line 47, in <module>
    from cam import ui, ops, curvecamtools, curvecamequation, curvecamcreate, utils, simple, \
  File "C:\Users\me\SourceCode\blendercam\scripts\addons\cam\ui.py", line 34, in <module>
    from cam import gcodeimportparser, simple
  File "C:\Users\me\SourceCode\blendercam\scripts\addons\cam\simple.py", line 31, in <module>
    from shapely.geometry import Point, LineString, Polygon, MultiLineString
ModuleNotFoundError: No module named 'shapely'
```

Interestingly, after all of this, I found that the packages were once again installed in `C:\Users\me\AppData\Roaming\Python\Python310\site-packages`

pppalain, 2023-05-16:
you are running the wrong  blender.  you say it is 3.41 but 3.3 is listed in the directory.

AVHon, 2023-05-16:
...I never said I was using Blender version 3.4. I'm using the latest LTS release, which is 3.3.6

AVHon, 2023-06-20:
I made some progress on this: I uninstalled Blender and deleted the folder `C:\Users\me\AppData\Local\Python`. Then I downloaded the portable version of Blender 3.3.8 LTS and extracted it to a folder in my home directory (`C:\Users\me\Software`). I followed the [blenderCAM zip file install instructions](https://github.com/vilemduha/blendercam/blob/master/documentation/Blendercam%20Installation.md#install-using-zip-file). This time, it worked! Rather than in the `AppData` folder, blender's pip put shapely in `C:\users\me\Software\blender-3.3.8-windows-x64\3.3\python\lib\site-packages`, which Blender apparently find it in just fine.

Kosovircek, 2023-07-20:
I have "solved" this problem by typing > **help('modules')**    into the **python interactive console** (scripting workspace inside blender) and it list all the modules. After that > import shapely    somehow works (i guess the help command searches a different place for it and caches it for other imports?)

pppalain, 2023-12-14:
hot fix was removed since users complained it took too long to load

closed 2023-12-14 by pppalain