vilemduha/blendercam #237: Crash using Blender 3.5.1

tirithen, 2023-05-23:
Hi! While trying out this interesting tool I have a crash when clicking the "Drawer" button:

```
AttributeError: Calling operator "bpy.ops.curve.simple" error, could not be found
```

It seems like the .simple API has been removed? I found this Q/A https://blender.stackexchange.com/a/276486 and there they switched to `bpy.ops.curve.primitive_bezier_curve_add`, not being so familiar with the Blender API, is that a viable alternative for newer Blender versions?

[image]

pppalain, 2023-06-05:
you need to activate the plugin curve extra it is included in blender but not activated

tirithen, 2023-06-05:
Thanks a bunch! I'll try that out next time I use the plugin!

closed 2023-06-05 by tirithen