vilemduha/blendercam #240: Modify plunge speed to independent number instead of percentage of feedrate

eastshores, 2023-06-15:
Original issue was raised regarding the time calculations and the plunge speed not being accounted for. Currently plunge speed is a percentage 1-100 of feedrate. This issue is to investigate changing this property to an independent number.

### Code analysis:
35 instances of variable plunge_feedrate across 33 files (two of which are definitions)

Defined in /scripts/cam/__init__.py as
line 764 under #feeds
plunge_feedrate: FloatProperty(name="Plunge speed ", description="% of feedrate", min=0.1, max=100.0, default=50.0, precision=1, subtype='PERCENTAGE', update=updateRest)
Defined in /scripts/__init__.py as
line 834 under #feeds
plunge_feedrate: FloatProperty(name="Plunge speed ", description="% of feedrate", min=0.1, max=100.0, default=50.0, precision=1, subtype='PERCENTAGE', update=updateRest)

Why is it defined twice?

Values are set for tool presets:
Value is set to 30.0 for all tool presets except Finishing_3mm_ballnose which is set to 50.0 (not sure why this one is different?)

Percentage value is set in gcodepath.py on line 293:
plungefeedrate = millfeedrate * o.plunge_feedrate / 100

plungefeedrate is only in gcodepath.py (1 definition and 2 references)

### Findings:
- To change this property in the UI the only change required is to change the float property subtype to "UNSIGNED" - see below screenshot
- Defaults in tool presets should be updated to an appropriate value.
- To change from percentage line 293 on gcodepath.py should be changed to:
- plungefeedrate = o.plunge_feedrate

[image]

pppalain, 2023-06-15:
possible problem is backward compatibility... old files will have to flag plunge speed <= 100 and adjust accordingly

eastshores, 2023-06-15:
Good point. Do you know if any blendercam release information is saved in the blend file? That would be the ideal way to handle it. I don't know if there would be any cases where someone wants a plunge speed lower than 100. If there is no release info that might be another issue to look at, it would be simple to implement and give us a lot of flexibility on updates like this.