vilemduha/blendercam #241: Implement user level preferences for blendercam

eastshores, 2023-06-15:
Related to discussions on improving UI by offering "levels" to show and hide UI complexity, this would benefit from the ability to have a persistent user preference so that each time the user works with blendercam, their UI preferences are in place. There are many other preferences, such as machine setup that would benefit from a persistent preference store.

Currently the recommended approach is to set all values and save a "template" blend file to start from. While this works, it is less intuitive to users than the concept of user preferences.

The proposal is to implement a preferences class to hold all user preferences and map that class to a JSON file located under /home/user/.config/blender/[blender_version]/scripts/addons/blendercam/user.preferences.json (subject to change) - this seems to be the most appropriate location for both blender and operating system conventions.

The logic for loading user preferences should only be triggered if the users .blend file has not been saved. This will allow preferences to be loaded for new files while preserving settings in users template files, or any settings that have been modified from the default preferences that would be specific to that .blend file.

Serialization should attempt to use reflection to map properties from the preferences class to JSON and back, this would allow the user preferences to be extended without modifying the serialization logic. Properties not found in an existing user preferences file should default to appropriate values. In the event that breaking changes are made, a migration strategy should be implemented to ensure that users do not get errors when upgrading from previous versions of blendercam that implement user preferences.

eastshores, 2023-06-16:
Here is a start to the conceptual model for user preferences. I have not started testing the recent UI updates so my understanding of the UI related preferences are based on conversation at this point.

[image]

eastshores, 2023-06-16:
preferences class has been created and serializes to json through __str__() .. when the user preferences class is initialized it calls LoadPreferences which checks for the existence of the path <user home>\Blender Foundation\Blender\3.5\config\addons\blendercam and creates that folder if it doesn't exist. It then checks for the user.preferences.json file in that location and creates the preferences based on default values if the file doesn't exist. Sample JSON content:

[image]

pppalain, 2023-12-14:
user level preferences were added in last pull.  Basic is very minimalist

closed 2023-12-14 by pppalain