vilemduha/blendercam #251: Autogenerate bridges fails on blender 4.0

greggjaskiewicz, 2023-12-04:
Python: Traceback (most recent call last):
  File "/opt/blender/blendercam/scripts/addons/cam/ops.py", line 748, in execute
    bridges.addAutoBridges(o)
  File "/opt/blender/blendercam/scripts/addons/cam/bridges.py", line 76, in addAutoBridges
    for c in curve.geoms:
AttributeError: 'list' object has no attribute 'geoms'

Seems similar to this: https://stackoverflow.com/questions/52131547/attributeerror-geoseries-object-has-no-attribute-geom

pppalain, 2023-12-08:
I can confirm that bridges work on blender 4.0

greggjaskiewicz, 2023-12-08:
Looks like the issue was on my part, well - partially. 
Picking wrong item in the object selector shouldn't result in an exception. I should get a dialog telling me off, or I shouldn't be able to pick wrong item in the first place - I think!

pppalain, 2023-12-14:
not an issue.  user mistake.  Training wheels and sanity checks may be needed.

closed 2023-12-14 by pppalain

phlplowe9, 2023-12-16:
So which object do I need to select?

Steps to reproduce:
open blender->general
Renderer->Cam
Interface->advanced
Select Default cube->cam operations->+
resize the cube to fit into the workspace.
Click auto generate bridges->error
click the default cube->auto generate bridges->error
click the machine wireframe->auto generate bridges->error
There's not a whole lot I can select and everything seems to generate the same error

abosafia, 2023-12-16:
Try bridges with a curve not mesh. We should fix that.