vilemduha/blendercam #255: Opencamlib is not installed

SUS316L, 2024-01-26:
Hi, I've just started using BlenderCAM with Blender 4.0 and experiencing an error "Opencamlib is not installed" displayed in “CAM info and warnings”.
"opencamlib_readme.txt" says it works with Python 2.7, but I believe Blender 4.0 is working on Python 3.1.
My question is: Is Opencamlib compatible with Blender 4.0?
If yes, which settings should I check?
I would appreciate it if you could help me.

abosafia, 2024-01-27:
Hi
There is a new way to install blendercam.
https://github.com/pppalain/blendercam/releases/tag/0.9.11
Follow instructions.

SUS316L, 2024-01-28:
Thanks for the advice, I reinstalled blender-4.0.2-windows-x64, installed blendercam.zip according to the instructions, but “CAM info & warnings” shows "BlenderCAM version: 0.9.11" and "Opencamlib is not installed". "Show experimental features" of "Scene: CAM – gcode generation tools" is turned on. Is there anything I should check?

SUS316L, 2024-01-30:
By manually copying the opencamlib folder in the blendercam.zip file to C:\Program Files\Blender Foundation\Blender 4.0\4.0\scripts\addons, the message "Opencamlib is not installed" no longer appears. I hope this will be helpful to those who are struggling with the same problem.

closed 2024-01-30 by SUS316L