vilemduha/blendercam #264: A groove is made narrower

denprog, 2024-06-25:
I want to make groove with the Parallel path. But it always is narrower after CNC than it is in Blender. I tried different options.
[image]
For example, this picture shows a groove and such path settings:
Strategy: Parallel
Distance between toolpaths: 2 mm
Distance along toolpaths: 0.2 mm
Use layers: 5 mm
Operation depth start: -1 mm
Movement type: Meander / Zig Zag
Cutter: End
Cutter diameter: 3 mm.
The picture shows that paths are too far from the walls of the groove.
What's wrong in my path? How to make a groove exactly as in Blender?

denprog, 2024-06-25:
I don't use skin. Are talking about this group? https://groups.google.com/g/blender-cam-forum

abosafia, 2024-06-25:
https://matrix.to/#/#blendercam:matrix.org

closed 2024-08-14 by pppalain