vilemduha/blendercam #268: Help wanted: "No module named `curve_simplify`

ParsaJafarian, 2024-07-28:
## Problem Summary
After manually installing some python packages (see issue #267), I got another module error:
[image]

This time, however, when I try to install it through pip by running `.\python.exe -m pip install curve_simplify`
 I get the following error:
```
ERROR: Could not find a version that satisfies the requirement curve_simplify (from versions: none)
ERROR: No matching distribution found for curve_simplify
```

I am not sure what to do now. Any help would be appreciated.

abosafia, 2024-07-30:
Try this 
https://github.com/pppalain/blendercam

pppalain, 2024-08-14:
blender 4.2 has made quite a few problems.  Blender makes updates and those break things for us

closed 2024-08-14 by pppalain