vilemduha/blendercam #271: error with a remove_doubles operator

luigipacheco, 2024-08-22:
Hello I have BlenderCAM running on windows with blender 4.2 so maybe it not compatible.

I get the following errors when using the curve tools:
AttributeError: Calling operator "bpy.ops.curve.remove_double" error, could not be found

I have installed all the required dependencies

ydnatag, 2024-08-27:
I had this error last week and I reported it to the maintainer: https://github.com/pppalain/blendercam/issues/166

TL;DR: Blender now does not include some extensions by default. Install `Simplify curves+` and `Extra Cruve`

abosafia, 2024-08-28:
Some updates to curve remove doubles on the unstable branch.
https://github.com/pppalain/blendercam

Kovadon, 2024-09-03:
Installed both `Simplify Curves+` and `Extra Curve` in `4.2` and `4.1` and am still experiencing this error

pppalain, 2024-10-25:
should be fixed with latest release

closed 2024-10-25 by pppalain

willpuckett, 2025-01-22:
I'm getting fhis error occasionally as well when trying to process a cam operation. I have both Simplify curves+ and Extra Curve installed...

[image]

willpuckett, 2025-04-01:
I'm getting 'context is incorrect' in object mode. 

It can usually be avoided by selecting an object and running the op again.

[image]