vilemduha/blendercam #275: Compatibility issue stemming from repos auto-updating python to 3.12

mj1911, 2024-12-15:
**Describe the Bug**
_Let your OS auto-update the system python from 3.11 to 3.12_

**System Information:**
 - OS: all which auto-update to python 3.12
 - Blender Version: 4.3.0
 - Addon Version: 1.0.57

**Error Message:**
_Extension bl_ext.user_default.fabex is incompatible (This Python version (3.12) isn't compatible with (3.11))_

mj1911, 2025-01-05:
Ok so, took matters into my own hands.

* Since system Python updates tend to purge added wheels, on Arch I had to `pamac install python-shapely` since it disappeared.
* I then extracted the latest .zip file, changed the names of the wheels from cp311 to cp312, fixed these in the manifest, and recompressed it.
* Can get file here: https://github.com/mj1911/misc/blob/main/BlenderCAM-fabexcnc-20250104.zip

It installs and seems to work.  But this very much is a hack and could end in flames, so anyone reading proceed at your own risk.  Ideally, wait for the devs to get everything updated properly instead.

kostrahb, 2025-03-13:
Same error here. Extension bl_ext.user_default.fabex is incompatible (This Python version (3.13) isn't compatible with (3.11)). Is there any possibility to make sure that this extension works in python 3.13?