vilemduha/blendercam #280: [BUG] Difficulty

willpuckett, 2025-01-22:
**Describe the Bug**
I am experiencing difficulty exporting chains. 

**To Reproduce**
Using a rectangular plane with two circles in it—

[image]

I create an outside profile cut operation for the plane, and an inside profile cut operation for the circles.

Running `Calculate Path & Export Gcode` on the operations individually outputs the tool paths and gcode file, but when I group them together as a chain, I get `AttributeError: Curve object has no attribute 'vertices'`.

[image]

I have noticed that when outputting an operation individually, the created path is a mesh:

[image]
[image]

But when I create an additional operation's path, the first path is changed into a curve:

[image]

I tried converting the curve path to a mesh and using the `Export Chain Gcode` button and the command executes without error, but when I cut it the travel moves didn't lift correctly and the workpiece was gouged in several places.

**Expected Behavior**
Pressing `Calculate Chain Paths & Export Gcode` will calculate all paths and export them without error, regardless of what is selected.

**System Information:**
 - OS: MacOS Sequoia
 - Blender Version: 4.3.2
 - Addon Version: 1.0.61

**Error Message:**
AttributeError: Curve object has no attribute 'vertices'

**Additional context**

willpuckett, 2025-01-22:
I'm also sometimes getting a remove doubles error when processing a cam operation, as in #271.

robinhyman, 2025-02-03:
I am also getting this exact issue running extension version 1.0.65 in Blender 4.2.2 LTS on Windows

pppalain, 2025-02-20:
this problem is being worked on

closed 2025-12-31 by pppalain