vilemduha/blendercam #282: [BUG] Extension bl_ext.user_default.fabex is incompatible. (This Python version (3.13) isn't compatible with (3.11))

jamincollins, 2025-01-29:
**Describe the Bug**
Extension bl_ext.user_default.fabex is incompatible. (This Python version (3.13) isn't compatible with (3.11))

**To Reproduce**
1. Install Arch Linux and Blender
2. Grab latest blendercam release
3. Try to install it

**Expected Behavior**
The blendercam release installs.

**Screenshots**

[image]

**System Information:**
 - OS: Arch Linux
 - Blender Version: 4.3.2
 - Addon Version: 1.0.61

**Error Message:**
Extension bl_ext.user_default.fabex is incompatible. (This Python version (3.13) isn't compatible with (3.11))

**Additional context**
None

blackbookpro, 2025-03-01:
I have the exact same problem with a different OS:
- OS: Fedora Asahi Remix (aarch64)
- Blender: 4.3.2
- FabexCNC: 1.0.65

Error: 
Extension bl_ext.user_default.fabex is incompatible (This Python version (3.13) isn't compatible with (3.11))

Please help!!!

blackbookpro, 2025-03-01:
Ok, after some research i have found a solution that seems to work!
there is a similar issue with CAD Sketcher and there somebody wrote a script to rename all the .whl files from 311 to 313 and to update the manifest.toml file too.

I adjusted that script and could now successfully install FabexCNC:

Just save it as modify_zip.py for example and run it with python3 in the folder where you have the fabexcnc.zip:

```
import tempfile
import zipfile
from pathlib import Path

# Open the source ZIP file
zip_path = Path("./fabexcnc.zip")
with zipfile.ZipFile(zip_path, "r") as zip_ref:
    # Create a temporary directory
    with tempfile.TemporaryDirectory() as tmp_dir:
        tmp_path = Path(tmp_dir)
        zip_ref.extractall(tmp_path)
        
        # Rename .whl files
        for whl in tmp_path.glob("**/*.whl"):
            new_name = whl.name.replace("cp311", "cp313")
            print(f"Renaming {whl.name} to {new_name}")
            whl.rename(whl.parent / new_name)
        
        # Update the manifest before creating the new ZIP
        manifest = tmp_path / "cam" / "blender_manifest.toml"
        if manifest.exists():
            manifest_str = manifest.read_text()
            manifest_str = manifest_str.replace("cp311", "cp313")
            manifest.write_text(manifest_str)
        
        # Create the new ZIP file
        new_zip_path = Path("./fabexcnc-py313.zip")
        with zipfile.ZipFile(new_zip_path, "w") as new_zip:
            for file in tmp_path.rglob("*"):
                if file.is_file():  # Ensure it's a file
                    arc_name = str(file.relative_to(tmp_path))
                    new_zip.write(file, arc_name)

print("New ZIP archive created successfully.")

```

mhedoc, 2025-09-01:
Same problem here:

System Information:

OS: CachyOS(Arch Derivate)
Blender Version: 4.5.2 LTS
Addon Version: 1.0.68
Installed python: 3.13

Error:
Extension bl_ext.user_default.fabex is incompatible (This Python version (3.13) isn't compatible with (3.11))

PabloBelmonte, 2026-02-22:
I arrived here from another addon, but the issue was the same. I adapted the script so it should work for any addon. Just change the variables FROM_VERSION, TO_VERSION and ZIP_NAME:

``` python
import tempfile
import zipfile
from pathlib import Path

FROM_VERSION = "311"
TO_VERSION = "314"
ZIP_NAME = "3dgs_render_by_kiri_engine_4.1.4"

# Open the source ZIP file
zip_path = Path(f"./{ZIP_NAME}.zip")
with zipfile.ZipFile(zip_path, "r") as zip_ref:
    # Create a temporary directory
    with tempfile.TemporaryDirectory() as tmp_dir:
        tmp_path = Path(tmp_dir)
        zip_ref.extractall(tmp_path)

        # Rename .whl files
        for whl in tmp_path.glob("**/*.whl"):
            new_name = whl.name.replace(f"cp{FROM_VERSION}", f"cp{TO_VERSION}")
            print(f"Renaming {whl.name} to {new_name}")
            whl.rename(whl.parent / new_name)

        # Update the manifest before creating the new ZIP
        for manifest in tmp_path.glob("**/blender_manifest.toml"):
            if manifest.exists():
                manifest_str = manifest.read_text()
                manifest_str = manifest_str.replace(
                    f"cp{FROM_VERSION}", f"cp{TO_VERSION}"
                )
                manifest.write_text(manifest_str)

        # Create the new ZIP file
        new_zip_path = Path(f"./{ZIP_NAME}-py{TO_VERSION}.zip")
        with zipfile.ZipFile(new_zip_path, "w") as new_zip:
            for file in tmp_path.rglob("*"):
                if file.is_file():  # Ensure it's a file
                    arc_name = str(file.relative_to(tmp_path))
                    new_zip.write(file, arc_name)

print("New ZIP archive created successfully.")
```